ut-robotics/picr21-team-tlo-rock #24: [Mechanics] - motor holder design tweaks

Akustav, 2021-11-06:
2 enhancements:
1. Reduce the radius of the holder, this makes creating it faster and wastes less material
2. Put the shaft fastening hole between the wheel fastening holes. This way there can be no interference bewteen the two fasteners

[image]

priitro, 2021-11-06:
1. Reduced motor holder radius to 28 mm. 
2. Positioned holes around, and now interference will be between two different fasteners.
[image]
Motor holder sketch 
[image]

Akustav, 2021-11-06:
Maybe we can make it 25? Then we can use 25mm materials

priitro, 2021-11-06:
Its is possible now - diameter is 25 mm and 2 mm from edge of hole to perimeter
[image]

Akustav, 2021-11-07:
Seems ok now

closed 2021-11-07 by Akustav