ut-robotics/picr21-team-tlo-rock #26: [Mechanics] - wheel could damage field elements or robots

Akustav, 2021-11-07:
When the wheel is spinning it could damage parts of the fields (baskets) or robots it hits with bolt heads. You you plan to cover the wheels? 

I not maybe recessed screws are better.

[image]

priitro, 2021-11-07:
1.I have designed wheel cover. I (we - others don't now yet about it ) haven't decided yet use it or not. If we use cover we can use same screws, if not i can change screws type and redesign holes. 
[image]
[image]

2. Bottom plate design is`'t ready yet and i don't know whether we use protective edge or not. Right now i have placed motors and wheels to bottom plate and we may have space for edge on it.
[image]

Akustav, 2021-11-07:
Okay, Seems like you have considered it. Will close

closed 2021-11-07 by Akustav

priitro, 2021-11-07:
I redesigned holes - now I can use both screws depending of whether I use cower or not:  Design of screw may be bad but it gives idea I mean.
[image]