ut-robotics/picr21-team-tlo-rock #33: Wrong motor driver package

ReikoR, 2021-11-24:
HVSSOP (28) package is used, but the available drivers are in VQFN-HR (14) package.

Package is also specified in the guide:
https://reikor.github.io/picr21-basketball-robot-guide/electronics/components.html#_motor_driver

JoelJaemaa, 2021-11-29:
New footprint has no thermal pad under it, I will draw that in manually

ReikoR, 2021-12-04:
Still wrong footprint and symbol.

Recommended (available in the lab) component:
https://reikor.github.io/picr21-basketball-robot-guide/electronics/components.html#_motor_driver
>DRV8243
>Product code: PDRV8243HRXYQ1 (HW variant)
>Package: VQFN-HR (14)

From the datasheet:
[image]

ReikoR, 2021-12-08:
It probably doesn't matter too much, but it would be nice to keep minimum solder mask expansion.

[image]

JLCPCB specifies the minimum at 0.05 mm.
https://jlcpcb.com/capabilities/Capabilities

[image]

[Datasheet ](https://www.ti.com/lit/ds/symlink/drv8243-q1.pdf) recommends 0.05 mm.

[image]

JoelJaemaa, 2021-12-13:
I can't seem to find this in the rules. There is a rule in design rules -> mask -> solder mask expansion -> expansion top and it's set at 4 mil.

ReikoR, 2021-12-14:
For this component it's defined on the component footprint. Component library needs to be changed.

closed 2022-01-07 by ReikoR