ut-robotics/picr21-team-tlo-rock #34: Incorrect microcontroller pins used

ReikoR, 2021-11-24:
USB and programmer signals can be connected only to specific pins. PWM output and encoder input signals can also be connected to limited amount of pins.

Recommended way for finding correct pins is to use Pinout & Configuration view in STM32CubeIDE.

More information can be found on the firmware page in the guide:
https://reikor.github.io/picr21-basketball-robot-guide/electronics/firmware.html

JoelJaemaa, 2021-11-29:
Updated the pins. Let me know if these are okay now

ReikoR, 2021-12-04:
Servo needs a separate timer as it needs different PWM frequency from motors and thrower. PB0 pin does not have any usable timer currently.
[image]
[image]

What interface is accelerometer using? I2C? PA5 and PA10 pins are not connected to anything usable in the MCU.
[image]

closed 2021-12-12 by ReikoR