ut-robotics/picr21-team-tlo-rock #36: Use 0603 package for resistors and smaller capacitors

ReikoR, 2021-11-24:
Most of the passive components in the lab are in 0603 package but some components on the schematic are in 0805 package. It doesn't make sense to order new components if the components with same values are available. For higher capacitances it's ok to use larger packages, but it's better to check what's already available in the lab.

JoelJaemaa, 2021-11-29:
Changed 0805 footprints to 0603

ReikoR, 2021-12-04:
There are still components in 0805 package.

[image]
[image]
[image]
[image]
[image]
[image]

JoelJaemaa, 2021-12-06:
Fixed

closed 2021-12-08 by ReikoR