ut-robotics/picr21-team-tlo-rock #40: Select correct resistors for DRV8243 configuration pins

ReikoR, 2021-11-24:
Some pins can be connected directly to GND, left unconnected or connected with specific resistor to GND.

For example currently ITRIP pin is connected to GND with a 0 Ohm resistor, which results in output current not being limited. I would recommended current limit of around 2 to 3 A. Current limit can be set with resistors connected to ITRIP and IPROPI pins.

[image]
[image]

Relevant information can be found from the datasheet:
https://www.ti.com/lit/ds/symlink/drv8243-q1.pdf

Let me know if more help is needed.

JoelJaemaa, 2021-12-06:
The resistors in the schematic are currently just placeholders so they could be placed on the pcb. I will update them soon.

JoelJaemaa, 2022-01-13:
Changed ITRIP resistor to 47K. Does that seem okay? Rest of them should be fine at 0 ohm (GND)?

ReikoR, 2022-01-14:
Resistor values should be fine, but place the resistor better on the schematic. Currently it's a bit hard to understand which is which.  It would be better to place resistors vertically.

[image]

JoelJaemaa, 2022-01-24:
Resistor placement fixed. Made other schematic elements nicer as well. Changed the values of feedback resistors in the switching regulator with values that were available in the lab. Same ratio. 5V is nice and stabile.

closed 2022-01-25 by ReikoR