ut-robotics/picr21-team-tlo-rock #43: Move game state handling into separate functions

ReikoR, 2021-11-28:
https://github.com/ut-robotics/picr21-team-tlo-rock/blob/f928aeb4cd5ac96405098a886fadbd8b1d98b824/robot%20code/game_logic.py#L76-L86

It would be good to avoid long list of nested if else statements.

peeter-virk, 2021-12-06:
done

ReikoR, 2021-12-10:
https://github.com/ut-robotics/picr21-team-tlo-rock/blob/8d38ac2402675cea33a5d87779682db3a0426ee8/robot%20code/game_logic.py#L177

Enum class is given as an argument to function instead of correct game state variable.

peeter-virk, 2021-12-11:
Fixed in main

ReikoR, 2021-12-13:
1.
This:
```python
if noball.value == 0:
    gs = GameState.moveto
    print("moveto")
    return gs
```

Can be written like this:
```python
if noball.value == 0:
    print("moveto")
    return GameState.moveto
```

In general it's better to not change function arguments.

2.
https://github.com/ut-robotics/picr21-team-tlo-rock/blob/b54554c1068070ccf952da579a595b7e8e827c08/robot%20code/game_logic.py#L129-L130

I don't think that `launchdelay` and `launch_time` are working, because those arguments are not mutable.

peeter-virk, 2022-01-01:
fixed

ReikoR, 2022-01-12:
1. There are more places where state can be returned without assigning it to `gs` variable:

https://github.com/ut-robotics/picr21-team-tlo-rock/blob/e6500201c194a8a11493a955380e087cac76c924/robot%20code/game_logic.py#L67-L70

https://github.com/ut-robotics/picr21-team-tlo-rock/blob/e6500201c194a8a11493a955380e087cac76c924/robot%20code/game_logic.py#L125-L127

https://github.com/ut-robotics/picr21-team-tlo-rock/blob/e6500201c194a8a11493a955380e087cac76c924/robot%20code/game_logic.py#L67-L70

2. Instead of making `launchdelay` and `launch_time` into arrays, new values for these variables could be returned by the `launch` function:

https://github.com/ut-robotics/picr21-team-tlo-rock/blob/e6500201c194a8a11493a955380e087cac76c924/robot%20code/game_logic.py#L197

```python
def launch(gs, launchdelay, speeds, delta, tgt, nearest_ball, launch_time):
    if launchdelay < 1:
        #...
        launchdelay += delta
    else:
        movement_vector = move_omni(5,0)
        #...
        launch_time += delta
        
        if launch_time > 6:
            launchdelay = 0
            launch_time = 0
            #...

    return gs, launchdelay, launch_time 
    
def main(nearest_ball, speeds, state, noball, basket):
    launchdelay = 0
    launch_time = 0

    while True:
        #...
        elif gs == GameState.launch:
            gs, launchdelay, launch_time = launch(gs, launchdelay, speeds, delta, basket, nearest_ball, launch_time)
```

peeter-virk, 2022-01-13:
fixed

closed 2022-01-14 by ReikoR