ut-robotics/picr21-team-tlo-rock #53: Use switching regulator for servo power

ReikoR, 2021-12-08:
Linear regulators waste a lot of power when currents are high and voltage difference is big.

JoelJaemaa, 2021-12-11:
Okay, I'll look into it but that linear regulator should work okay for now. It will dissipate more heat but it has a small load and the batteries should be able to handle it. If I have time I will replace it on monday but if not then can we keep it this way? I have the part, no need to order it.

ReikoR, 2021-12-12:
Some recommendations:

1. You can use Texas Instruments Webench to generate the design for the regulator:
https://www.ti.com/design-resources/design-tools-simulation/webench-power-designer.html
Other manufacturers have similar tools available.

2. Check that the regulator is in stock:
https://www.mouser.ee/c/power/power-management-ics/voltage-regulators-switching-regulators/?m=Texas%20Instruments&number%20of%20outputs=1%20Output&topology=Buck&instock=y
There are also some regulators in the lab from last year.

It's not that the batteries are not able to handle it. Linear regulators waste a lot of power. The load from servo is not that small.

JoelJaemaa, 2021-12-13:
Okay. I'll do this tonight.

JoelJaemaa, 2021-12-13:
I'll come by the lab tomorrow and check what we have. Everything the TI Webench recommended is out of stock or on order. I know that a linear regulator is not a good option but I put it in initially because it's what I have. I also do stuff for Estcube so I'll check if we have anything in stock there that I could get. Other than moving stuff from M1 layer and this regulator issue, everything else should be better now.

ReikoR, 2021-12-15:
I ordered these regulators:
https://www.ti.com/product/TPS62933

You can use those or the ones you find in the lab.

I'll probably order these inductors also:
https://ee.farnell.com/bourns/srp6060fa-5r6m/inductor-shld-5-6uh-10a-aec-q200/dp/2901832?st=2901832

JoelJaemaa, 2021-12-15:
Found a couple TPS54227s in the lab. Will use those. I did not check what size inductors we have though.

JoelJaemaa, 2021-12-15:
Schematics and board have been updated with TPS54227 based buck converter

ReikoR, 2021-12-15:
1. Vias in TPS54227 footprint are not connected to GNDB.

[image]

Footprint has pad 9, but symbol does not have pin 9, so the pad is not connected to anything.

Also it's better to add vias in PCB design not in component footprint. Then it's easier to change via sizes and locations.

2. Inductor used right now is in 1206 (3216 metric) size. I don't think there are any inductors in that size that can meet the recommended specs. TI Webench recommends: L: 4.7 μH - 7.05 μH, DCR: 125 mΩ (max), IDC: 3.64 A (min).

3. More input capacitance should be used. Current values are below the recommendations. It's probably better to use higher capacitance than recommended, because the servo motor could have high peak current draw.

JoelJaemaa, 2021-12-17:
I'd like to change it but this is a downloaded library and not one that I made. So the vias are already there. I'd like to remove the vias and add pin 9. Also edit the other footprints to move stuff around between mechanical layers. How can I do that?

I added more capacitance to the input.

ReikoR, 2021-12-18:
You can open the existing library files and modify them or you can create a new library yourself. It's easy to create components with Tools -> IPC Compliant Footprint Wizard.

JoelJaemaa, 2021-12-23:
Pin 9 GNDB issue fiexd

JoelJaemaa, 2021-12-23:
Proper inductor also used now

ReikoR, 2021-12-27:
1. Which exact inductor are you using in the design? Footprint in the design is for Würth Elektronik 744777930, but that one is not suitable. Würth has 4.7 uF inductors in similar package.

2. Move inductor closer to regulator.

Recommendation from [TPS54227 datasheet](https://www.ti.com/lit/ds/symlink/tps54227.pdf):

> 2\. Keep the SW node as physically small and short as possible to minimize parasitic capacitance and
inductance and to minimize radiated emissions. Kelvin connections should be brought from the output to the
feedback pin of the device.

[image]

3. Avoid routing anything below the inductor.

[image]

4. Output capacitance is still below the recommendations.

From TI Webench:

Cin = 20uF (2x 10uF)
Cout = 44uF (2x 22uF)

[image]
[image]

JoelJaemaa, 2021-12-29:
Using a 4.7uH inductor I found in the lab. Has the same footprint. Using this Würth part for the footprint.

JoelJaemaa, 2021-12-29:
Added more capacitance and shortened L1 path.

JoelJaemaa, 2021-12-31:
All issues should be fixed now

ReikoR, 2022-01-01:
This has not been fixed:

3\. Avoid routing anything below the inductor.

[image]

JoelJaemaa, 2022-01-02:
Fixed

ReikoR, 2022-01-02:
[image]

Linear regulator is not needed anymore, since the switching regulator can provide the 5V supply.

JoelJaemaa, 2022-01-03:
Removed it and put all the battery side 5V stuff onto the switching regulator

ReikoR, 2022-01-07:
Vias in TPS54227 footprint are again not connected to GNDB. It's better to remove vias from footprint and add them in PCB design.

[image]

JoelJaemaa, 2022-01-07:
Yeah it does not like to stay connected. Removed them from footprint and added manually. Should be fixed now.

closed 2022-01-07 by ReikoR