ut-robotics/picr21-team-tlo-rock #57: Connect decoupling capacitors as close as possible to the components

ReikoR, 2021-12-08:
[image]

Track inductance should be minimized by keeping them as short as possible, otherwise decoupling capacitors won't be as effective.

JoelJaemaa, 2021-12-11:
Fixed

ReikoR, 2021-12-12:
Capacitors need to be much closer.

[image]

JoelJaemaa, 2021-12-13:
Fixed

ReikoR, 2021-12-15:
Decoupling capacitors for motor driver VM pins could be be closer to the drivers by putting capacitors to bottom layer and connecting them with vias.

[image]

JoelJaemaa, 2021-12-17:
Fixed. Also added an extra 4.7uf cap there. Moved both to bottom.

closed 2021-12-27 by ReikoR