ut-robotics/picr21-team-tlo-rock #60: Avoid routing ground

ReikoR, 2021-12-08:
[image]

Ground should be connected to internal ground layer(s) with vias as close as possible to component pads.

JoelJaemaa, 2021-12-11:
Fixed

ReikoR, 2021-12-12:
Vias should be as close as possible to pads to reduce inductance from tracks.

[image]

JoelJaemaa, 2021-12-13:
Fixed the long grounds

ReikoR, 2021-12-15:
This via could still be closer:
[image]

Why are you using the default via size everywhere? Vias can be smaller and moved closer to the pads.

Design rule for via sizes can be changed:
[image]

JoelJaemaa, 2021-12-17:
Used a smaller via and placed it closer. 
I'm using default size because I did not necessarily need to use any other size before.

closed 2021-12-27 by ReikoR