ut-robotics/picr21-team-tlo-rock #61: Use wider tracks or polygons for higher currents

ReikoR, 2021-12-08:
[image]

JoelJaemaa, 2021-12-11:
I tried making some improvements. Hopefully it should be a bit better now.

ReikoR, 2021-12-12:
It's better, but you should move the tracks a bit and use smaller vias (at least for signals) to make power tracks for drivers wider. 

Driver's VM capacitors should also be as close to the driver as possible. SMD capacitors can also be on the bottom of the board and connected with vias.

[image]

JoelJaemaa, 2021-12-13:
Will do it tonight

JoelJaemaa, 2021-12-13:
I managed to get the caps a bit closer and unchecked the option that removes loops from traces. This allowed me to get more vias on one trace and get the motor driver power connections connected from both sides.

closed 2021-12-27 by ReikoR