ut-robotics/picr21-team-tlo-rock #62: Board outline should be drawn on Mechanical 1 (M1) layer

ReikoR, 2021-12-08:
Currently it's on top layer, which is for top layer copper.

There are some other things on M1 layer, which should be moved to some other layers.

[image]

Component bodies are typically placed on M13 (Top 3D Body) or M14 (Bottom 3D Body) layers.

JoelJaemaa, 2021-12-11:
Moved outline to M1. Can't move other stuff on M1 to M13 or M14. It only lets me choose top, bottom or the two middle GND layers. Can I move them to one of these?

ReikoR, 2021-12-12:
Other stuff needs to be moved in component libraries.

JoelJaemaa, 2021-12-13:
Will fix tonight

JoelJaemaa, 2021-12-23:
I moved the stuff in libraries but it does not seem to update on the PCB. Even after deleting and re-adding the component. Can you help me with this, please? Otherwise it should all be done. Regulator stuff is fixed as well now.

ReikoR, 2021-12-27:
You probably need to use **Tools -> Update From PCB Libraries...** to update the footprints.

JoelJaemaa, 2021-12-31:
Should be fixed now

closed 2022-01-01 by ReikoR