ut-robotics/picr21-team-tlo-rock #70: Very confusing variable naming

Akustav, 2021-12-14:
Variable names like noball is quite bad for the readability of the code. Its concept is described in this one comment line and later on used.

https://github.com/ut-robotics/picr21-team-tlo-rock/blob/b54554c1068070ccf952da579a595b7e8e827c08/robot%20code/main.py#L65

When I look here it seems like a integer from 0 to 1 that works as a boolean:
https://github.com/ut-robotics/picr21-team-tlo-rock/blob/b54554c1068070ccf952da579a595b7e8e827c08/robot%20code/game_logic.py#L58-L60

But below it is:
https://github.com/ut-robotics/picr21-team-tlo-rock/blob/b54554c1068070ccf952da579a595b7e8e827c08/robot%20code/game_logic.py#L67-L71

Overall very unclear as to what the noball variable does.

peeter-virk, 2021-12-17:
It is time since last ball was detected or the time no ball has been in any camera frame.

ReikoR, 2022-01-12:
Variable name should be changed to something that is easier to understand. For example `no_ball_time`.

peeter-virk, 2022-01-13:
Contrary to a common belief longer variable names do not make the code more readable. There are reasons that comments exist. Besides, it isn't such a big issue  if it was, it should have been brought up earlier. Changing variable names now is just a great way to introduce bugs into the code which will only cause more problems down the line.

ReikoR, 2022-01-14:
Problem is not with the length of the name. Short names are only better if they are able describe the purpose of the variable. This issue was created because it is hard to understand what the variable does.

IDEs have refactoring tools to help rename variables:
https://code.visualstudio.com/docs/editor/refactoring#_rename-symbol
https://www.jetbrains.com/help/pycharm/rename-refactorings.html

You can even use GitHub to find where it is used:
https://github.com/ut-robotics/picr21-team-tlo-rock/search?q=noball

peeter-virk, 2022-01-17:
fixed it

closed 2022-01-17 by ReikoR