ut-robotics/picr21-team-tlo-rock #72: Check and fix design rule errors

ReikoR, 2021-12-15:
For example in current design:

[image]

JoelJaemaa, 2021-12-17:
That was an accidental reduntant trace from moving stuff. Will look over them and fix.

ReikoR, 2022-01-01:
1. **Short-Circuit Constraint Violation** and **Hole To Hole Clearance Constraint Violation**

Remove holes inside the slots in USB connector footprint. These are not needed.

[image]

2. **Hole Size Constraint Violations**

Increase Maximum value in HoleSize rule.

[image]

[JLCPCB Capabilities](https://jlcpcb.com/capabilities/Capabilities):
Drill Hole Size: 0.20mm- 6.30mm

JoelJaemaa, 2022-01-02:
Fixed

closed 2022-01-02 by ReikoR