ut-robotics/picr21-team-tlo-rock #73: Multiple combine operations

Akustav, 2021-12-16:
https://github.com/ut-robotics/picr21-team-tlo-rock/blob/640eadb84083cdc3f1cb29fb544fb14cbe74ae40/robot%20code/game_logic.py#L110-L112

Here you combine rotations determined by the closest ball multiple times. This is expected? Why can it not be done with one combine?

peeter-virk, 2021-12-17:
Combine takes in only two movements can be changed but not high on the priority list.

ReikoR, 2022-01-12:
What is the reason for using `rotate_omni` two times in a row?

peeter-virk, 2022-01-13:
commented the second one out

ReikoR, 2022-01-14:
What was the purpose of the second line? You should probably delete it, if it's not needed.

peeter-virk, 2022-01-16:
It was a leftover from some testing I did. Was no longer supposed to be active.

closed 2022-01-17 by ReikoR