ut-robotics/picr21-team-tlo-rock #75: Use tented vias

ReikoR, 2021-12-27:
Using tented vias might make the soldering a bit easier by helping to avoiding the solder from flowing into vias.

Also might help to avoid shorts when vias are close to different component pads. There does not seem to be any vias like that in the design.

Probably the easiest option to use tented vias, is to create new Solder Mask Expansion rule for vias:

[image]

Vias under the the driver's are quite big and might make soldering harder even when being tented. It might be better to use small (0.2/0.45) vias, that don't pull away so much solder from the pads.

[image]

JoelJaemaa, 2021-12-29:
Added this rule

ReikoR, 2022-01-01:
Can't see the rule and vias not tented either:

[image]

JoelJaemaa, 2022-01-02:
Should be fixed now. Last time Altium crashed while saving and looks like some changes were not saved.

closed 2022-01-02 by ReikoR