ut-robotics/picr21-team-tlo-rock #79: Add stoppers to top of battery holder

Akustav, 2022-01-02:
I would add a stopper to the top of the holder. Additionally i would remove the bottom tab since it can make it hard to operate:

[image]

priitro, 2022-01-03:
Removed bottom tab. Now its may be not necessary to add stopper to top? 
[image]

Akustav, 2022-01-08:
Maybe just in case round the corners off, then it is harder to cut into a battery and damage it?

[image]

priitro, 2022-01-08:
Made arc  cut 
[image]

priitro, 2022-01-08:
[image]

closed 2022-01-08 by Akustav