ut-robotics/picr21-team-tlo-rock #81: Add supports to thrower internal ramp

Akustav, 2022-01-02:
I suggest adding some cutouts to the top plate of the chassis. This reduces the possibility that the ramp bends. the ramp there is quite thin.:
[image]

priitro, 2022-01-03:
Added supporting top plate body.  Between ramps it is lower because if it`s in a same distance it would resist free ball movement. Throwing motor squeeze's ball between ramps.  Right now distance is 3,55mm
[image]

[image]

Akustav, 2022-01-08:
I would suggest making the dogbones something like this: 
[image]

Then the ramp is better supported in the direction most forces will apply to it.

priitro, 2022-01-08:
Made corrections:
[image]

closed 2022-01-08 by Akustav