ut-robotics/picr21-team-tlo-rock #89: Top plate seems difficult to remove

maidlam, 2022-01-09:
How is the top plate removal planned?
The top plate seems to be unable to move in any direction other than up and down a few mm, and unremovable without removing the thrower or ramps.
[image]

Because of the battery box, the plate can't be removed at an angle.

One way of insertion would be to extend the cutout, and have the NUC and ramp top bolts removed for the duration of the top plate insertion.
This would also require extending the wires of the power switches and NUC power socket.
[image]

priitro, 2022-01-09:
Unfortunately planning that top plate design, the thrower`s ramps height never come to mind. Its not good!

Made following changes in design:
*changed throwers throwing angel from 58 to 65 degrees. It gives 1mm cut off from thrower height, with regular thrower more than 5 mm.
*lifted thrower and ramp middle holding bolt 5mm
*lifted camera 5mm
*lifted NUC  and connectors holes  ca 10 mm
*increased cutoff of the camera and NUC stand 28 mm
Now it gives 5.4 mm space  (10.25-4.8)  for plate insertion and removal.
-unfortunately  ramps upper polts must remove before. 
[image]

For regular thrower with throwing angle 65 degrees, maybe ramp don't need upper bolts for supporting.
Space for insertion-removal top plate is 14.9 mm (19.7-4.8mm).
[image]
[image]

priitro, 2022-01-13:
If I turned the batteries to a narrower side, the NUC could no longer be switched on. Battery box seen with new dimensions [image]
[image]

closed 2022-01-25 by ReikoR